weiss19ja/amos-ss16-proj2 #5: Save all logging output to file

m-hofmann, 2016-05-18:
Right now only communication logs are stored in a file. However, @Maximilian-Fischer and @andreas-bauer would like to have a file containing all logging output produced by the backend application.

uv48uson, 2016-05-19:
im not sure whether this is the best solution. keep in mind that we have the intention to show the logging file to the admin-user later in the project. therefore it might be easier to have all the logging in one file!? or at least clearer

m-hofmann, 2016-05-19:
Please clarify what you mean. This task is about storing all logging output in one file and additionally having a communications log.

uv48uson, 2016-05-19:
ah ok i see. Nevertheless it might be useful to have all (logs and communication) in one file. At least if the backend logs contain somehow interesting information for the user!?

m-hofmann, 2016-05-19:
I think this is more of a developer related issue.

uv48uson, 2016-05-19:
Probably. Sry then.

commit 7dc479e3 (2016-05-21): Iss-05 log to file all.log

Fixes #5
closed 2016-05-21 by m-hofmann