weiss19ja/amos-ss16-proj2 #8: Head movement is delayed

vwiedemann, 2016-05-22:
The first time the head movement is triggered, it occasionally takes some seconds to trigger the movement. Only happens on first movement, probably a problem with the startup-process.

vwiedemann, 2016-05-23:
The delayed head movement is caused by the socket being opened up too late. The socket opens around 5 seconds after loading the driving mode page.
This is probably caused by the setup-process being triggered too late.

Maximilian-Fischer, 2016-05-30:
I will work on this bug with andreas this week.I already created a branch for this bug. Our first approach will be to open the socket connection already in the main page.

m-hofmann, 2016-05-30:
FYI: we could only reproduce this issue with Firefox on @vwiedemann 's system. Chrome works fine.

Maximilian-Fischer, 2016-05-30:
OK I just recognized that the rover connection gets established pretty fast (less than half a second) with firefox on my developer machine.

m-hofmann, 2016-05-30:
I think we should close this issue then.

closed 2016-05-30 by m-hofmann

andreas-bauer, 2016-05-30:
On my system with chrome browser connection established immediately.