weiss19ja/amos-ss16-proj2 #21: Async frontend tests are broken

m-hofmann, 2016-06-12:
Asynchronous test cases should be written in the form 

    it('should async do something', function(done) {
      // do
      setTimeout(function() {
        // expect something
        done(); // notify about finished test case
      }, 250);
    });

Currently, test cases for roverDrive and roverService leave out the `done` callback. This makes them seem pass - but the function within `setTimeOut` is never called. However, when trying to add additional test cases, the test suite will run longer so that the callback from `setTimeout()` gets executed eventually. As the frame for the test has been destructed by then, jasmine will throw an error about unknown spy methods and the tests fail.

When trying to fix the test cases by writing them as noted above, they fail. Especially the Promise-based construct in roverService seems fragile to test.

Maximilian-Fischer, 2016-06-14:
Fixed in commit 1aa50e38372008fa956962521a162b0d12f30576 and merged to master in commit 1046b93b4d0f039df37e3eddd3f0e676d6b22890. Implemented in collaboration with Andreas Bauer.

m-hofmann, 2016-06-14:
Problem still exists in `roverDriveSpec.js`. The spy on `enterDriverMode()`/`exitDriverMode` is not being called when refactoring the test case to use the `done` callback as shown above.

closed 2016-06-26 by Maximilian-Fischer