weiss19ja/amos-ss16-proj2 #24: Missing License Headers in Source Files

andreas-bauer, 2016-06-15:
Here are my recommendation for a license header.

/**
 * This file is part of Mobile Robot Framework.
 * Mobile Robot Framework is free software under the terms of GNU AFFERO GENERAL PUBLIC LICENSE.
 */

/**
 * This file is part of Mobile Robot Framework.
 * Mobile Robot Framework is free software under the terms of GNU AFFERO GENERAL PUBLIC LICENSE.
 *
 * Included third-party software components:
 * ComponentXY, License GPL
 */

m-hofmann, 2016-06-18:
I'd go with option one, but extend it somewhat to match the example [here](http://grepcode.com/file/repo1.maven.org/maven2/org.neo4j.maven/licenses/1/AGPL-3-header.txt).

The second one is far too tedious as we'd have to update the headers when adding a new component.

weiss19ja, 2016-06-23:
Added the first licence header suggestion of andreas-bauer to all file types which are supporting comments.

Fixed with commit 59271e6e02c461226d00fb966136a85d5abfa954

closed 2016-06-23 by weiss19ja