weiss19ja/amos-ss16-proj2 #36: Both rovers have the same wifi SSID

vwiedemann, 2016-07-02:
Both rovers use 192.168.0.1 which might cause interference problems (and 192.168.0.1 is also the default address for wifi routers so it might cause problems with them as well)
Solution: Change SSID

m-hofmann, 2016-07-02:
One Rover has the SSID ROVER the other has the SSID ROVER_NEO. As you can only be connected to one Wi-Fi at a time (usually, edge cases ignored), the rover's IP address should not cause a collision.

vwiedemann, 2016-07-02:
I wasn't intending to write SSID but IP. Sorry for confusion.

m-hofmann, 2016-07-02:
Yeah, but as you are only connected to one Wi-Fi at a time I really don't see this as a problem. How is it broken?

vwiedemann, 2016-07-02:
When I use the desktop it always crashes because the LAN connection also uses 192.168.0.1 for the router. Therefore all Commands that are supposed to go to the rover go to the router and fail. As most of the wifis have 192.168.0.1 as a standard IP, I guess it might fail not only at my place ;)

m-hofmann, 2016-07-02:
I don't think that being connected to the rover and another device is a real use case for us. But we should as @uv48uson or @MarkusSchmitz  for clarification.

m-hofmann, 2016-07-09:
Fixed on both rover's. MK I has IP address `192.168.23.1`, MK II has IP address `192.168.42.1` adjustment was documented in the [wiki](https://github.com/weiss19ja/amos-ss16-proj2/wiki/Configuring-the-wireless-hotspot#operating-with-different-ip-addresses).

closed 2016-07-09 by m-hofmann