weiss19ja/amos-ss16-proj2 #40: Unable to change from observer mode to rover master

vwiedemann, 2016-07-02:
When being in observer mode with desktop pc, clicking on "Rover Master" doesn't trigger rover master mode. This occurs only from time to time. Client's Console gets spammed with
TypeError: t.options.zone is null
 nipplejs.min.js:1:11477
image loaded roverDrive.js:91:25
init joystick for id zone_joystick_2 joystickService.js:38:7
TypeError: t.options.zone is null
 nipplejs.min.js:1:11477
image loaded roverDrive.js:91:25
init joystick for id zone_joystick_2 joystickService.js:38:7

Rover console doesn't show any message at all

m-hofmann, 2016-07-03:
Seems to be Firefox-only.

Maximilian-Fischer, 2016-07-09:
Fixed with commit c0d9e82e17757a30038c69cdf6497e13c517125d

closed 2016-07-09 by Maximilian-Fischer