weiss19ja/amos-ss16-proj2 #43: Nullpointer exception in doSendNotification to client

vwiedemann, 2016-07-02:
Wanted to access Logs with only one connected client, got the following exception:

Exception in thread "Thread-33" java.lang.NullPointerException
        at de.developgroup.mrf.server.ClientManagerImpl.doSendNotificationToClient(ClientManagerImpl.java:225)
        at de.developgroup.mrf.server.ClientManagerImpl.notifyClientById(ClientManagerImpl.java:185)
        at de.developgroup.mrf.server.controller.LoggingCommunicationControllerImpl.sendResponseToClient(LoggingCommunicationControllerImpl.java:117)
        at de.developgroup.mrf.server.controller.LoggingCommunicationControllerImpl.sendLogEntriesToClient(LoggingCommunicationControllerImpl.java:70)
        at de.developgroup.mrf.server.controller.LoggingCommunicationControllerImpl.lambda$getLoggingEntries$0(LoggingCommunicationControllerImpl.java:41)
        at java.lang.Thread.run(Thread.java:745)

vwiedemann, 2016-07-03:
As this exception blocks all the interactions with the rover afterwards, I think this bug is more important than prioC

weiss19ja, 2016-07-03:
Fixed issue with commit d68f5c0ad24a949263b0e6648675e9f6373bd195

closed 2016-07-03 by weiss19ja