weiss19ja/amos-ss16-proj2 #54: After some time of ideling, a "badMessage" message appears

vwiedemann, 2016-07-02:
"I had one device connected (Android, Chrome and the log view was open, after some time of doing nothing I got this in the backend console and the server shut itself down:

2016-06-13 23:07:37,345 WARN  (qtp9861382-22) [HttpParser(parseNext:1347)] badMessage: 400 Illegal character 0x16 in state=START in '\x16<<<\x03\x01\x00\x00>>>generate_204 HTTP...\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00' for HttpChannelOverHttp@1f7807f{r=0,c=false,a=IDLE,uri=-}"

m-hofmann, 2016-07-02:
This seems to be related to the Android Connectivity Check as we can see from the string `generate_204`. The string relates to the URL Android devices try to call to forward users to portal start pages on pay for use WiFi hotspots: [see this Stackoverflow post](http://android.stackexchange.com/questions/123129/how-does-wifi-in-android-detect-if-the-device-has-to-sign-in-or-not).

weiss19ja, 2016-07-09:
It seems that the issue is occured by an illigeal character in the clients browser which is not parseable by the jetty httpparser. [See this Stackoverflow post for possible causes.](http://stackoverflow.com/questions/25625410/jetty-9-warning-badmessage-400-illegal-character)

With the commit e0ce839d25146b50275c937403bca3ca6e1ed57d we excluded the jetty httpparser from the log file.

closed 2016-07-09 by weiss19ja